hasenbanck/remram #5: Write build instructions for RemRam 1.1.

hasenbanck, 2018-07-27:
We need to write Wiki entries for the build process of a RemRam v1.1 board with pictures and debug information.

gauix4, 2018-10-25:
Hello
so if I understand correctly we can use the whole series of STM32F7
I have plenty of STM32F767ZIT6

hasenbanck, 2018-10-25:
No, you can't use the STM32F767ZIT6 instead. It's 144 pin, whereas the STM32F765VIT6 the design is based on has 100 pins.

The variant definition in the Ardunio core is also based on the STM32F765VIT6, so you would need to write a whole new variant if you want to use another MCU (which would need to be pin / configuration compatible). And the Marlin stuff has also STM32F765VIT6 specific stuff.

So no ... you really can't use a STM32F767ZIT6 instead.

gauix4, 2018-10-25:
it exists in version also LQFP100 is what I have ^^

hasenbanck, 2018-10-25:
No, the STM32F767ZI is the model number for LQFP144. LQFP100 for the STM32F767 has the part number STM32F767VI and STM32F767VG.

If you mean those two, then the answer is still: No, you can't just replace them. You would need to check the configuration with STM32Cube MX (the current configuration is provided with the sources) to make sure that it's really pin / configuration compatible.

Then you would also need to write your own board definition file for the Arduino core. And you would need also to adjust the Marlin sources to support your MCU with this board.

If you are willed to do that, then you could try doing it. But it's not supported by this design and the current software (Core and Marlin).

gauix4, 2018-10-25:
So I will still order the STM32F765VIT6 ^^

hasenbanck, 2018-10-25:
Yes, please stick to the BOM.

gauix4, 2018-10-25:
I could always help develop the project a bit more

hasenbanck, 2018-10-25:
Help is always welcome, but RemRam starting from 1.3 will stay backward
compatible. So you would need to keep that in mind.

The software side only lacks the CDC support in the core and I already made
a merge request for that. But of course it could need more work.

V1.3 is also not tested yet well. So there could be a v1.4 if people find
bugs in the design.

I personally will only support it. My main development resources as of
right now move into the heteromycin project of mine that you can also find
on GitHub, were I take a more radical approach from the lessons I learned
with RemRam. You are also welcome to help there.