hasenbanck/remram #12: Writing to SD card overCDC/VCOM doesn't work

hasenbanck, 2019-01-12:
Currently we can't write to SD card using pronterface/pronsole. It will try to create/open a file on the SD card and will fail doing. Reading from a onboard SD card works fine.

Things we tried but didn't fix the issue:
 * Lower the baud rate of the CDC / VCOM port
 * Lowering the SPI speed
 * Using either an external or the onboard sd card reader

This bug needs further testing / investigation to find the possible cause.

hasenbanck, 2019-01-12:
Further testing revealed:
 * Saving a file over UART to SD card also doesn't work
 * Using M928 (saving a logfile to SD card) also doesn't work

Maybe the error lies in the marlin HAL for the STM32.

hasenbanck, 2019-01-12:
@ktand Is the M928 working on your ARMED board using an external SD card reader? i want to know this, because if it is working, then my board design would be the cause. If it doesn't, it would mean the cause lies inside the STM32 HAL inside Marlin.

ktand, 2019-01-12:
I can check later today.

hasenbanck, 2019-01-12:
@ktand Thank you very much!

ktand, 2019-01-12:
@hasenbanck Doesn't seem to work on my board either, so I guess there is some more work to be done in the STM32 Marlin HAL. I can however print from SD card but not upload files to it.

hasenbanck, 2019-01-12:
@ktand Thank you for testing. I'm a little bit relieved that it's not hardware related and that it seems to be the STM32 Marlin HAL. I will poke around and will look for the issue in the next days.

And yes, reading works fine on my end too. SPI writing also works, since my TMC2130 drivers are getting configured. So it seems to be related to either the block writing of data or maybe the FAT filesystem code (?).

ktand, 2019-01-13:
@hasenbanck I have identified the problem and created a PR, see https://github.com/MarlinFirmware/Marlin/pull/12896

ktand, 2019-01-13:
@hasenbanck Made an error in my previous Marlin PR... Created a new one: https://github.com/MarlinFirmware/Marlin/pull/12897

hasenbanck, 2019-01-13:
@ktand Thank you very much! You saved ma a lot of time.

leseaw, 2019-01-17:
@ktand @hasenbanck  fix isnt working :( worse than before, now not even reading possible.

hasenbanck, 2019-01-17:
@leseaw Rebased my config to the newest bugfix 2.0 master. Seems to work fine on my end for the reading. Writing still seems bugged after some kilobytes, but that is still under investigation like already said some days before.

hasenbanck, 2019-01-17:
Changing the SPI mode from 0 to 3 solved all of my writing problems. I will catch some sleep and think about this over the weekend.

hasenbanck, 2019-01-19:
Changing to SPI mode 3 shouldn't be necessary, since all devices on the bus are mode 0 devices. I changed back the mode 0 and measured the MOSI, MISO and SCLK lines. The resistance of the MOSI and MISO lines were 27k, so it seem that the STM32 also has PU activated even when not running. I removed the extra PU to remove some variables in the testing. The SPI speed is also greatly reduced while testing.

I then used my oscilloscope to measure the SPI bus (Blue= SCLK, RED = MOSI, YELLOW = MISO):

[image]
[image]
[image]

These are clearly a SPI mode 0 signals. The timing of the signal also seems fine. There is an odd spike before writing a byte on the MOSI line (that seem to bleed in the SCLK line).

I couldn't measure the burst writing (that is failing) yet. But if these odd spikes are there while bursting, this could explain the errors we are seeing right now. I will try to add some capacitor values that could act as a filter. I'm also not sure where these spikes could originate. The TMC drivers are deactivated while testing this bug.

hasenbanck, 2019-01-19:
Here is a capture with a 1nF on the MISO and MOSI line (pretty aggressive). I tested this with one quarter SPI speed. Full speed won't work since the filtering would be too aggressive. The burst writing seems clean and fine. 

[image]

I observed some off spikes on the CLK line.

[image]

No clue were this spikes come from, if it could count as "HIGH" for the SD card and why this should or even could influence only the SD writing.

I try to capture the SPI bus traffic with a logic analyzer next.

hasenbanck, 2019-01-20:
Attached a logic analyzer and took a dump of the write operation to SD card (Pulseview):

[pulseview.zip](https://github.com/hasenbanck/remram/files/2776477/pulseview.zip)

Overall the dump seems fine. The SCLK, MISO and MOSI register cleanly and it seems the little "noise spikes" shown above are not a problem (at least for the logic analyser). I don't have a deep insight into the SPI based SD card protocol, so I can't comment on that yet. I also don't know how well the SPI SD card protocol is implemented into Pulseview.

I will try to add a very small capacitor on SCLK line to filter out those little spikes, just to be safe. Would be great if there would be other F7 designs in Marlin for test against, but the only other design is currently dormant.

Those little spikes for now are my best target to investigate further. After I filtered them out and the SD card writing shouldn't work, I will go deeper into the overall SD card operation / protocol. The random nature of the problem and the clean output that it produces (the gcode fragments that are written on the card are clean and don't contain any ASCII garbage as you might expect on a noise/corrupt bus) make it rather hard to pinpoint a likely culprit.

hasenbanck, 2019-01-22:
A user on discord debugged the problem separately and came to the conclusion, that the noise spike comes from initializing the SPI device again and again when doing byte write transactions by Marlin.

The spike is the CLK line pulling up on SPI device re-init. The "big" spikes are happening, when there is an interrupt firing while the SPI device is initializing: Then it will have enough time to get pulled up to logic high.

A PD resistor on the CLK line of 47k seem to work (tested by the other user, but not by me yet). I will have to try it myself, maybe this is a good workaround for now.

I don't know why Marlin is calling ```spiInit()``` as often as it does right now. But it seems that at least on my device (STM32F765VI) this isn't really a great idea, since the SPI CLK line behavior described above (since the CS for the currently active device is asserted high).

So besides adding a weak PU (47k) there's maybe the need to add a fix in the STM32 HAL. Following options spring to my mind:

 1. Prevent Marlin to call spiInit() multiple times. Maybe we store the current configuration and only apply a new configuration, if it differs from the old one.
 2. Disable interrupts when inside the spiInit() function. But I don't know if this would be enough for all cases.

@ktand I don't know why you are not affected from this behavior. Maybe the F4 behaves differently on SPI reinit (or maybe you have a PD on the CLK line, but I doubt that).

hasenbanck, 2019-01-23:
47k is too low. 4.7k is needed as a PD.