hasenbanck/remram #14: Marlin will not compile for RemRam

myval, 2020-05-14:
Hi,
Just got RemRam V1 board and have been trying to compile Marlin 2.0 and/ or 2.1dev but no luck.
I get these errors:
`
Marlin\src\HAL\STM32_F4_F7\HAL_SPI.cpp: In function 'void spiRead(uint8_t*, uint16_t)':
Marlin\src\HAL\STM32_F4_F7\HAL_SPI.cpp:125:9: error: 'class SPIClass' has no member named 'dmaTransfer'; did you mean 'transfer'?
  125 |     SPI.dmaTransfer(0, const_cast<uint8_t*>(buf), nbyte);
      |         ^~~~~~~~~~~
      |         transfer
Marlin\src\HAL\STM32_F4_F7\HAL_SPI.cpp: In function 'void spiSendBlock(uint8_t, const uint8_t*)':
Marlin\src\HAL\STM32_F4_F7\HAL_SPI.cpp:157:9: error: 'class SPIClass' has no member named 'dmaSend'
  157 |     SPI.dmaSend(const_cast<uint8_t*>(buf), 512);
Compiling .pio\build\STM32F7\src\src\HAL\shared\eeprom_i2c.cpp.o
      |         ^~~~~~~
Compiling .pio\build\STM32F7\src\src\HAL\shared\eeprom_spi.cpp.o
*** [.pio\build\STM32F7\src\src\HAL\STM32_F4_F7\HAL_SPI.cpp.o] Error 1
In file included from e:\documents\github\marlin-2.0.x\marlin\src\hal\hal.h:26,
                 from e:\documents\github\marlin-2.0.x\marlin\src\inc\marlinconfig.h:30,
                 from Marlin\src\HAL\shared\backtrace\../../../core/serial.h:24,
                 from Marlin\src\HAL\shared\backtrace\backtrace.cpp:29:
Marlin\src\HAL\shared\backtrace\backtrace.cpp: In function 'bool UnwReportOut(void*, const UnwReport*)':
e:\documents\github\marlin-2.0.x\marlin\src\hal\stm32_f4_f7\hal.h:55:21: error: 'SerialUART1' was not declared in this scope; did you mean 'SerialLP1'?
   55 |   #define MYSERIAL0 SerialUART1
      |                     ^~~~~~~~~~~
Marlin\src\HAL\shared\backtrace\../../../core/serial.h:62:40: note: in expansion of macro 'MYSERIAL0'
   62 |   #define SERIAL_OUT(WHAT, V...) (void)MYSERIAL0.WHAT(V)
      |                                        ^~~~~~~~~
Marlin\src\HAL\shared\backtrace\../../../core/serial.h:88:27: note: in expansion of macro 'SERIAL_OUT'
   88 | #define _CHAR_1(c)        SERIAL_OUT(write, c)
      |                           ^~~~~~~~~~
Marlin\src\HAL\shared\backtrace\../../../core/serial.h:86:27: note: in expansion of macro '_CHAR_1'
   86 | #define __CHAR_N(N,V...)  _CHAR_##N(V)
      |                           ^~~~~~
Marlin\src\HAL\shared\backtrace\../../../core/serial.h:87:27: note: in expansion of macro '__CHAR_N'
   87 | #define _CHAR_N(N,V...)   __CHAR_N(N,V)
      |                           ^~~~~~~~
Marlin\src\HAL\shared\backtrace\../../../core/serial.h:99:27: note: in expansion of macro '_CHAR_N'
   99 | #define SERIAL_CHAR(V...) _CHAR_N(NUM_ARGS(V),V)
      |                           ^~~~~~~
Marlin\src\HAL\shared\backtrace\backtrace.cpp:38:3: note: in expansion of macro 'SERIAL_CHAR'
   38 |   SERIAL_CHAR('#'); SERIAL_PRINT(*p, DEC); SERIAL_ECHOPGM(" : ");
      |   ^~~~~~~~~~~
*** [.pio\build\STM32F7\src\src\HAL\shared\backtrace\backtrace.cpp.o] Error 1`

I have searched the net but no luck to find solution.
Thanks
Josef

myval, 2020-05-14:
> 
Marlin\src\HAL\STM32_F4_F7\HAL_SPI.cpp: In function 'void spiRead(uint8_t*, uint16_t)':
Marlin\src\HAL\STM32_F4_F7\HAL_SPI.cpp:125:9: error: 'class SPIClass' has no member named 'dmaTransfer'; did you mean 'transfer'?
  125 |     SPI.dmaTransfer(0, const_cast<uint8_t*>(buf), nbyte);
      |         ^~~~~~~~~~~
      |         transfer
Marlin\src\HAL\STM32_F4_F7\HAL_SPI.cpp: In function 'void spiSendBlock(uint8_t, const uint8_t*)':
Marlin\src\HAL\STM32_F4_F7\HAL_SPI.cpp:157:9: error: 'class SPIClass' has no member named 'dmaSend'
  157 |     SPI.dmaSend(const_cast<uint8_t*>(buf), 512);
Compiling .pio\build\STM32F7\src\src\HAL\shared\eeprom_i2c.cpp.o
      |         ^~~~~~~
Compiling .pio\build\STM32F7\src\src\HAL\shared\eeprom_spi.cpp.o
*** [.pio\build\STM32F7\src\src\HAL\STM32_F4_F7\HAL_SPI.cpp.o] Error 1
In file included from e:\documents\github\marlin-2.0.x\marlin\src\hal\hal.h:26,
                 from e:\documents\github\marlin-2.0.x\marlin\src\inc\marlinconfig.h:30,
                 from Marlin\src\HAL\shared\backtrace\../../../core/serial.h:24,
                 from Marlin\src\HAL\shared\backtrace\backtrace.cpp:29:
Marlin\src\HAL\shared\backtrace\backtrace.cpp: In function 'bool UnwReportOut(void*, const UnwReport*)':
e:\documents\github\marlin-2.0.x\marlin\src\hal\stm32_f4_f7\hal.h:55:21: error: 'SerialUART1' was not declared in this scope; did you mean 'SerialLP1'?
   55 |   #define MYSERIAL0 SerialUART1
      |                     ^~~~~~~~~~~
Marlin\src\HAL\shared\backtrace\../../../core/serial.h:62:40: note: in expansion of macro 'MYSERIAL0'
   62 |   #define SERIAL_OUT(WHAT, V...) (void)MYSERIAL0.WHAT(V)
      |                                        ^~~~~~~~~
Marlin\src\HAL\shared\backtrace\../../../core/serial.h:88:27: note: in expansion of macro 'SERIAL_OUT'
   88 | #define _CHAR_1(c)        SERIAL_OUT(write, c)
      |                           ^~~~~~~~~~
Marlin\src\HAL\shared\backtrace\../../../core/serial.h:86:27: note: in expansion of macro '_CHAR_1'
   86 | #define __CHAR_N(N,V...)  _CHAR_##N(V)
      |                           ^~~~~~
Marlin\src\HAL\shared\backtrace\../../../core/serial.h:87:27: note: in expansion of macro '__CHAR_N'
   87 | #define _CHAR_N(N,V...)   __CHAR_N(N,V)
      |                           ^~~~~~~~
Marlin\src\HAL\shared\backtrace\../../../core/serial.h:99:27: note: in expansion of macro '_CHAR_N'
   99 | #define SERIAL_CHAR(V...) _CHAR_N(NUM_ARGS(V),V)
      |                           ^~~~~~~
Marlin\src\HAL\shared\backtrace\backtrace.cpp:38:3: note: in expansion of macro 'SERIAL_CHAR'
   38 |   SERIAL_CHAR('#'); SERIAL_PRINT(*p, DEC); SERIAL_ECHOPGM(" : ");
      |   ^~~~~~~~~~~
*** [.pio\build\STM32F7\src\src\HAL\shared\backtrace\backtrace.cpp.o] Error 1

hasenbanck, 2020-05-14:
I don't actively develop for REMRAM anymore, so you are essentially on your own.

From what I can see in the logs your provided you are using the wrong HAL. The REMRAM board ist supposed to use the ```HAL/STM32``` HAL and not the ```HAL/STM_F4_F``` HAL.  Be sure to use the offical Arduino core:
https://github.com/stm32duino/Arduino_Core_STM32

Good luck.